v923z/micropython-builder #4: ESP32

A622266, 2022-10-03:
Can you consider to build a ulab firmware for ESP32 and ESP32-S3?

The boards I have are:
ESP32_DevkitC_V4
ESP32-S3-DevkitM-1 v1.0

I'm following the directions to do this myself as well, but I am stuck on the final step of creating and running the make file.
For either of the above boards, can you elaborate on how to create the make file?  The part of the document I am trying to work through is here:

[image]

v923z, 2022-10-03:
> Can you consider to build a ulab firmware for ESP32 and ESP32-S3?

Yes, absolutely.

> The boards I have are: ESP32_DevkitC_V4 ESP32-S3-DevkitM-1 v1.0
> 
> I'm following the directions to do this myself as well, but I am stuck on the final step of creating and running the make file. For either of the above boards, can you elaborate on how to create the make file? 

Simply open a text editor, add 

```bash
BOARD = GENERIC
USER_C_MODULES = $(BUILD_DIR)/ulab/code/micropython.cmake

include Makefile
```
and save it under `ULAB_MICROPYTHON/micropython/ports/eps32/makefile`, where `ULAB_MICROPYTHON` is the folder, where you have `micropython`, and `ulab` side by side.

A622266, 2022-10-03:
I created the file at ./ULAB_MICROPYTHON/micropython/ports/esp32/makefile,
and added the 3 lines listed above as the contents to the file makefile.
I note among the many *.c and *.h files and other file types in the esp32
directory, there is another file called MakeFile
While in the esp32 directory, I run  the command make .
But my output is:
make: Nothing to be done for '.'.

Any idea what I am doing wrong?

Also, how to distinguish the firmware for my esp32 versus my esp32s3?  I
assume BOARD=GENERIC does something that should work with most esp32,but is
there way to make it more specific (not sure I really need that, but to
know how it is possible might help me to better understand the build
process)

v923z, 2022-10-05:
> I created the file at ./ULAB_MICROPYTHON/micropython/ports/esp32/makefile, and added the 3 lines listed above as the 

Here you need a directory name, not `ULAB_MICROPYTHON`. You should substitute here the name of your directory. So, e.g., in my case I have

```
artifacts  micropython  micropython-lib ulab
```
in a directory called

```
/home/v923z/sandbox/test
```
so you would create 

```
/home/v923z/sandbox/test/micropython/ports/esp32/makefile
```
Basically, 

```
/home/v923z/sandbox/test
```
is the directory, where you cloned both `micropython` and `ulab`.
 
I hope this helps.

A622266, 2022-10-06:
Yes I started over again and this time I completed the build process.  My error in the last step was executing 'make .' instead of 'make' (note period).  Oops!

I used BOARD=GENERIC_S3 in the makefile, since my ESP32 is an S3 model, and the build completed without error.

Now I am trying to flash the board over the USB at port /dev/ttyACM0, which I have confirmed to be the correct port.  I am using the command:
idf.py -p /dev/ttyACM0 flash
Unfortunately I got an error:  No serial data received.  
Maybe this isn't something you can help with, but if you have an idea, I would be glad to know.

A622266, 2022-10-06:
One thing I see is that the esptool.py version used automatically for the flash process is esptool.py v3.2-dev.  The latest esptool.py version is 4.3.  Do you think this should be updated?  The reason I am confident that the port and physical hardware connection is correct is that I can still run 'mpremote a0' and connect to the esp32s3.

A622266, 2022-10-06:
I was able to establish the connection.  It was a hardware issue.  However, I could not successfully flash the device. Following make, the created executable command fails since it recognizes the board as an S3. This may be because the S3 is flashed from memory 0x0, whereas older esp32 boards are flashed from 0x1000.

v923z, 2022-10-07:
You might have to edit the partition table: https://github.com/v923z/micropython-ulab/#what-to-do-if-the-firmware-is-too-large

A622266, 2022-10-07:
Yes there are few things that I believe must be changed.  I haven't found the right list or combination yet.  But I think there at least these three:
1)  IDF tools version has to be at least v4.4.  
2)  In makefile, BOARD should be set equal to GENERIC_S3.  I don't think this also has to be set in Makefile, but perhaps it does.
3) Flash should start from location 0x0 rather than 0x1000, which differs from other ESP devices.  You hint about the partition relates to this.

There may need to be something else to let the idf.py tool know that the target is an esp32s3 and not an esp32.  I just notice in some of the messages during the build process that an esp32 is assumed as the default.

A622266, 2022-10-16:
I succeeded in building ulab for my esp32 S3.  
1)  I used IDF tools version v4.4.  
2) In makefile, BOARD should be set equal to GENERIC_S3.

I did not have to make any other changes.

v923z, 2022-10-17:
Great! Would you mind adding your build script here?

A622266, 2022-10-21:
**Build script for ESP32 S3**
First, clone the `ulab`, the `micropython`, as well as the `espressif` repositories:
```py
export BUILD_DIR=$(pwd)

git clone https://github.com/v923z/micropython-ulab.git ulab
git clone https://github.com/micropython/micropython.git

cd $BUILD_DIR/micropython/

git clone -b v4.4 --recursive https://github.com/espressif/esp-idf.git
```
Then install the `ESP-IDF` tools:
```py
cd esp-idf
./install.sh
. ./export.sh
```
Next, build the `micropython` cross-compiler, and the `ESP` sub-modules:
```py
cd $BUILD_DIR/micropython/mpy-cross
make
cd $BUILD_DIR/micropython/ports/esp32
make submodules
```
At this point, all requirements are installed and built. We can now compile the firmware with `ulab`. In `$BUILD_DIR/micropython/ports/esp32` create a `makefile` with the following content:
```py
BOARD = GENERIC_S3
USER_C_MODULES = $(BUILD_DIR)/ulab/code/micropython.cmake

include Makefile
```
Now you can run `make`.

closed 2022-10-21 by A622266