v923z/micropython-builder #5: Raspberry Pi Pico W and ulab

sakluk, 2023-02-15:
I guess that [the nightly Releases](https://github.com/v923z/micropython-builder/releases) doesn't yet support the Wireless version of Raspberry Pi Pico?

I tried that maybe two weeks ago and it didn't reckognize the basic network module calls shown in [Getting started with Raspberry Pi Pico W](https://projects.raspberrypi.org/en/projects/get-started-pico-w/2)

That would be nice addition. Thank you!

sakluk, 2023-02-16:
> Päivämäärä: 16.2.2023 22.18 (GMT+02:00)
> Saaja: v923z/micropython-builder ***@***.***>
> Kopio: Sakari Lukkarinen ***@***.***>, Author ***@***.***>
>
> I would be more than happy to add this to the build chain. Do you want to create a build script for that?

I'll try. Never done such thing. Yesterday I learned how to build own version of micropython for Pico and Pico W.

Heading to that ... :-D

Here is a link to what I and my collagues are currently working for

https://github.com/sakluk/first_year_hardware_project

Best Regards,

Sakari

Lähetetty laitteesta Galaxy

-------- Alkuperäinen viesti --------

v923z, 2023-02-16:
> I'll try. Never done such thing. Yesterday I learned how to build own version of micropython for Pico and Pico W. Heading to that ... :-D 

But if you can build locally, then it shouldn't be hard to adapt the process to github. I'm sure there would be interest in that.

> Here is a link to what I and my collagues are currently working for https://github.com/sakluk/first_year_hardware_project

Very interesting. Thanks for sharing!

closed 2023-02-21 by v923z