v923z/micropython-builder #20: support ESP32-UNICORE variant

mzhboy, 2023-10-13:
This variant is supported on [micropython esp32 generic page](https://micropython.org/download/ESP32_GENERIC/)
can you please include this variant in micropython builder

v923z, 2023-10-14:
I would be happy to add this, but it's not clear to me, what the board's name is, and I don't find it in the micropython github workflow. https://github.com/micropython/micropython/blob/516385c4ccbb1f7fa9dbd37a44ed8ae33dcf1942/ports/esp32/boards/ESP32_GENERIC/board.md mentions that `UNICORE` is the board's name, but with that the script actually fails.

mzhboy, 2023-10-15:
It's a [generic esp32-solo-01 module](https://www.espressif.com.cn/en/news/new-espressif-module-esp32-solo-1)
 from espressif

v923z, 2023-10-15:
That doesn't compile: https://github.com/v923z/micropython-builder/actions/runs/6522719443/job/17712584072?pr=21. What I don't know is which board one has to specify for the build. 

In the PR https://github.com/v923z/micropython-builder/pull/21, you can see that I've tried quite a few variants, and all failed.

mzhboy, 2023-10-15:
Hi, after looking at [micropython building manual of esp32]($ make BOARD=ESP32_GENERIC BOARD_VARIANT=OTA).
I think the correct command should be 
`$ make BOARD=ESP32_GENERIC BOARD_VARIANT=UNICORE`
 board variant can be found at [esp32 board.json](https://github.com/micropython/micropython/blob/master/ports/esp32/boards/ESP32_GENERIC/board.json)

v923z, 2023-10-23:
Merged into master, closing now.

closed 2023-10-23 by v923z