v923z/micropython-builder #23: Firmware for Pimoroni Pico W and ulab

ipiriyan2002, 2023-11-27:
Hey, I am new to this type of work and was wondering if there is a framework for Pimoroni Pico W and lab. I need it for my project and am unsure as to how to do it since I am unfamiliar with C.

Thank you

v923z, 2023-12-02:
I think what you have to do is check, whether your MCU is mentioned herehttps://github.com/pimoroni/pimoroni-pico/blob/main/setting-up-micropython.md. If so, then you look at the workflow files in https://github.com/pimoroni/pimoroni-pico/tree/main/.github/workflows, and see what's under the section corresponding to your MCU.

ipiriyan2002, 2023-12-02:
The method I followed included forking the pimoroni pico repo and then adjusting the picow cmake file to include ulab and remove some sensors I did not need. So I am not sure I can really contribute a script per say since it was more of less using the workflow they had in place to generate the uf2 file.

v923z, 2023-12-03:
That is pretty much what would be required here. If you look at the scripts in this repository, they're really simple, most of them just two lines of bash.

ipiriyan2002, 2023-12-03:
[image]
So I am wondering how I would create the bash file if I had manually changed the pico cmake file they already had in a forked repo's custom branch. Something like above. How would I tell the system to look at my forked repo's custom branch?

Again, apologies, I am not the most proficient with C or GitHub actions