v923z/micropython-builder #24: ESP32 in WSL2 environment

callimero, 2023-12-08:
Hey, great tool! I already build a firmware for the Pico with it. 

System is Win10 with WSL(2) with Ubuntu as guest.

I now try to do that same for a ESP32 board. And I am stuck.

I come to 

`
...
Project build complete. To flash, run this command:
/home/cw/.espressif/python_env/idf5.0_py3.8_env/bin/python ../../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build-ESP32_GENERIC/bootloader/bootloader.bin 0x8000 build-ESP32_GENERIC/partition_table/partition-table.bin 0x10000 build-ESP32_GENERIC/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    23056  (    5616 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1885120  (  146496 remaining)
total                  1950656
make: Leaving directory '/home/cw/micropython-builder/micropython/ports/esp32'
`

I changed the (PORT) argument and the path to esptool.py

I get then:

`
cw@DESKTOP-PCKKJN6:~/micropython-builder$ /home/cw/.espressif/python_env/idf5.0_py3.8_env/bin/python ./esp-idf/components/esptool_py/esptool/esptool.
py -p /dev/ttyACM0 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0
x1000 build-ESP32_GENERIC/bootloader/bootloader.bin 0x8000 build-ESP32_GENERIC/partition_table/partition-table.bin 0x10000 build-ESP32_GENERIC/microp
ython.bin
usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]] [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'build-ESP32_GENERIC/bootloader/bootloader.bin'
`
So I am as it seems in a wrong path after the command.

`
cw@DESKTOP-PCKKJN6:~/micropython-builder$ ./scripts/esp32/generic.sh
`
idf.py says:

`
cw@DESKTOP-PCKKJN6:~/micropython-builder$ python3 ./esp-idf/tools/idf.py -p /dev/ttyACM0 flash
Please use idf.py only in an ESP-IDF shell environment.
cw@DESKTOP-PCKKJN6:~/micropython-builder$`

Help :) (I am quit out of my comfort zone here)

v923z, 2023-12-11:
Do you want to add this board to the github workflow here?

callimero, 2023-12-12:
It is a plain generic esp32 board, so it should work. I think I made some mistakes and the idf is not happy. 

I did a restart not using micropython-builder (installed idf in the system and followed the instructions on the mp github) and that works :(

BTW: With a RP2 Board (Arduino RP2040 Nano Connect) it worked just fine.

v923z, 2023-12-12:
OK, but then it should be in the regular releases, re-compiled every two days: https://github.com/v923z/micropython-builder/releases

F-Bello, 2024-04-01:
Hi all, I am also looking for the firmware to run on an esp32 board (Arduino Nano ESP32) without customisation. For me the releases page looks like this:
[image]
Apologies if I am missing something obvious (I am also out of my depth here!), but I'm not sure which one of those files I should use? Or should I be navigating to some other page altogether? 
Thanks!

v923z, 2024-04-01:
The ESP32 build is broken at the moment, so no artifacts are uploaded to the release page. I'll try to sort it out. If you want to look at it yourself, here is where you would start: https://github.com/micropython/micropython/blob/master/.github/workflows/ports_esp32.yml

F-Bello, 2024-04-01:
Ah ok, thanks for letting me know!