v923z/micropython-builder #28: RP2350 build?

cuedcad, 2024-08-27:
`Apologies if I have missed it, but will you be adding a RP2350/Pico 2 build that
makes use of the fpu in the CortexM33`?

Many Thanks to the other versions!

v923z, 2024-08-27:
You mean the new chip, right? micropython doesn't seem to have a build script for that chip, so it would be a bit hard for me to include it here. But I would be happy to add it.

cuedcad, 2024-08-28:
Yes the new mcu from RPi. As you have probably seen there is a 'preview' uf2 (https://micropython.org/download/RPI_PICO2/) but reading around there may be issues with the gpio pins and [https://www.youtube.com/watch?v=_O3Aeyv2ILM](https://www.youtube.com/watch?v=_O3Aeyv2ILM) seems to indicate that there may need to be some care taken when compiling to get the 
best performance, but hopefully on the near future. Thank you again for your builds

v923z, 2024-08-28:
OK, thanks for the clarification! I would suggest that we keep the issue open, and pick it up again, once micropython officially supports the chip. If I happen to forget, just ping me!

Derfies, 2024-10-22:
@v923z I'm also very keen to make use of this.

Are these the scripts you're referring to? Looks like there was some activity last week:

https://github.com/micropython/micropython/tree/master/ports/rp2/boards/RPI_PICO2

v923z, 2024-10-22:
Thanks for the pointer, I'll try to find some time to fix this.

v923z, 2025-05-18:
I haven't yet had time to deal with this.

Jay-Damodaran, 2025-08-11:
I've been trying to compile the micropython firmware for RP2350 for a few weeks now. After running all the commands listed in the main micropython-ulab github repo, no .uf2 files are being generated. Some attention to this issue would be appreciated. Here are the commands I am running. I've also tried several variants of this including calling git submodule update --init for just the modules listed in the tutorial on the main repo. I've also tried using cmake instead of make in the last step although cmake doesn't seem to recognize RPI_PICO2_W when passed in with the -DPICO_BOARD flag, which is weird. 

git clone https://github.com/micropython/micropython.git
cd micropython
git submodule update --init --recursive
cd ..
git clone https://github.com/v923z/micropython-ulab ulab

cd micropython/mpy-cross
make -j4
cd ../ports/rp2
make BOARD=RPI_PICO2_W submodules
make BOARD=RPI_PICO2_W clean
make BOARD=RPI_PICO2_W USER_C_MODULES=../../../ulab/code/micropython.cmake

These lines give me the following error upon entering the build folder created inside ../ports/rp2.

 *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory 'PiPico/micropython/ports/rp2/build-RPI_PICO2_W'
See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:65: all] Error 1

Jay-Damodaran, 2025-08-15:
No, it gives the same error. I wonder if it has to do with the picotool that's being used. Make gives a warning about not finding a picotool and then downloads it automatically from somewhere, but it mentions rp2040 when it's being downloaded.

Jay-Damodaran, 2025-09-10:
I think this issue can be closed now.