v923z/micropython-builder #30: Installation fails under Arch Linux

smalacar, 2024-11-20:
Hi guys, 

Thanks for sharing your scripts!
I encountered a problem while running the following script

`./scripts/esp32/generic-c3.sh`

I am using an arch-based distribution and when running the script, the espidf toolchain is not installed.
The script fails with an output like this: 

`./scripts/esp32/esp32.sh: line 12: esp-idf/export.sh: No such file or directory`

(The espidf toolchain is not installed at all, thats why the script is not found ^^)
I assume the problem lies in the `install_espidf.sh` file, because there some python packages are installed using apt. 

I totally understand that you may not want to adapt your scripts for other package managers. Maybe you can add a disclaimer that the builder can only be used with distros using apt. 

Cheers to you

v923z, 2024-11-20:
I would actually be happy to add support for arch, so if you think that you can install the packages through pacman, then we could just branch off before the package manager is called. This could benefit others, too.

v923z, 2025-08-16:
Since there's been no movement on this issue, and since it is strictly speaking not related to `ulab` as such, I'm closing it now. However, as I said above, I'd be happy to support `arch`, should someone want to implement the required changes.